ut-robotics/picr22-team-jackbot #5: [Mechanics] how do you do the sidewalls?

Akustav, 2022-10-05:
The sidewalls are 10mm thick, how do you plan to do them. 3D printing them can take a long time but it can be also milled.

You can split it to 2 4.8mm thick sheets that are bolted together in a few spots. Something like this:

[image]

Akustav, 2022-10-09:
There is still some issues here. The main one is that it is difficult to understand how the thrower mounts using the side extrusions.
Are these holes supposed to align?

[image]

I would suggest making the sidewalls and mounting posts as separate components that can be attached to each other. This would look something like this:

Red lines are the block and blue lines are holes through it that are used to mount it to the sidewall.
[image]

WDYT?

karlrahn, 2022-10-09:
I asked, if we can dril some holes to the LeBot, and we were allowed when I asked Reiko. In general we will make new holes for the robot.

karlrahn, 2022-10-12:
Made new pillars for walls, separated from walls. Full build will be made with printing with 3D printers, in total of 8 pieces. Some larger parts, like the ramps can be made with CNC.

Akustav, 2022-10-12:
This solution seems a lot simpler to make. I would mill the sidewalls and internal ramps and 3D-rpint the pillars. 3D milled pieces should be 4.8mm thick.

karlrahn, 2022-10-12:
I was thinking if the wall width can be made 4.5mm, because I saw that the stock version is 4.8mm

karlrahn, 2022-10-12:
In general, probably I will be milling the walls with width of 4.5mm using 4.8 material

closed 2022-10-16 by Akustav