ut-robotics/picr22-team-jackbot #10: [Mechanics] Thrower motor too high

Akustav, 2022-10-09:
The motor is too high off the ground, making it inoperable.

[image]

Akustav, 2022-10-12:
Interesting idea overall. But there is a issue:

It can not be lowered enough to touch the ball:
[image]

The middle ramp will touch the ground before contact can be done with the ball. I suggest shortening the middle ramp a little, This stuff can be cut:

[image]

I would suggest to set up the height of the thrower in a static way. There clearance of the robot should not change and therefor adjustability is not too important and can make the solution more fragile. If you insist on this solution, I suggest beefing up this area. Overall thicker walls there are usefull and I would add a fillet to the internal corners for extra strength:

[image]

karlrahn, 2022-10-12:
But if i make the sliding area's lenght longer, then it should be operable ?

Akustav, 2022-10-12:
You still need to make the small ramp shorter because the ramp can not go into the floor. With the current configuration the motor needs to go down around 13mm but the thrower is at most 12mm off the ground.

karlrahn, 2022-10-12:
It won't go into ground, if I don't fully lower the ramp.

karlrahn, 2022-10-12:
I can leave it to somewhere midpoint etc.

closed 2022-10-16 by Akustav