ut-robotics/picr22-team-jackbot #18: [Mechanics] How is the beautyrim attached

Akustav, 2022-11-02:
How is the beautyrim attached to the robot? There is currently no clear indication. Since it blocks some of the mounting holes it must also be removable.

Take into consideration that it should not break on impact.

karlrahn, 2022-11-03:
made a small pin which fill tightly fit into the hole on the wheel center. Just pull out or push in, no any screws or glue needed.

Akustav, 2022-11-08:
The main issue is that if another robot hits the wheel, will it stick there? Friction fits degrade over time so it will become looser and looser over time. Is this something to worry about?

karlrahn, 2022-11-21:
Fixed the beautyrim, actually made a new one, which is attatchable with the same bolts as for the wheels

closed 2022-11-27 by Akustav