ut-robotics/picr22-team-jackbot #20: [Mechanics] tweaks to motor mount

Akustav, 2022-11-02:
Some suggestions for the motor mount:

1. Remove the back support, it is not needed to support the back of the motor. Besides being a extra piece it can make assembly more difficult because you need to align all pieces to their corresponding holes.
2. How do you plan to manufacture the motor mounts? They can currently not be manufactured easily using 3D printing or milling.

Some ways to fix it:

- Don't have the motor at an angle, making the mounts simpler
- Have a two piece construction, where the angle is dictated by a separate mounting piece. Something like this:
[image]

Any other ideas are also welcome

karlrahn, 2022-11-13:
Removed the angle, no degrees

maidlam, 2022-11-21:
Back supports are still in the model. Do you want to keep them?

karlrahn, 2022-11-21:
I'll leave them, the new robot will be stiffer that way

karlrahn, 2022-11-22:
Also made the dogbones bit better on the upperplate and the motor mounts, added bolt holes under the lowerplate, to lock the  motors onto the lower plate.

closed 2022-11-27 by maidlam