ut-robotics/picr22-team-jackbot #29: [Mechanics] Incorrect cutout for switches

maidlam, 2022-11-21:
The switches have an incorrect cutout, the clips at the sides of the switches wont hold them in place.

Refer here for the correct cutout for the switches
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/design.html#_switches

karlrahn, 2022-11-21:
Fixed the cutouts for switches

maidlam, 2022-11-27:
The current cutouts will be very tight for the switches.
Make the cutout based on the image below (from mechanics guide).
The 2,5 mm at the bottom of the cutout is made so that the prongs of the switch will engage and fix the switch in place.

[image]

karlrahn, 2022-11-27:
So you suggest that I make it a bit looser?

karlrahn, 2022-11-27:
Made them looser

maidlam, 2022-11-29:
Change the bottom height of the cutout to 2,5 mm. That part is important for the switches.
The 11,1 mm and 14,4 mm parts are still not correct and increase the radius to 3,7 mm.

karlrahn, 2022-11-29:
I left the holes on the smaller cut, but changed the measurements now correctly. Should be ok I guess.

closed 2022-11-30 by Akustav