ut-robotics/picr22-team-jackbot #33: [Mechanics] Motor mount tenons don't have much support

maidlam, 2022-11-21:
The holes for the motor mount tenons don't have much support at the shorter sides.

[image]

One way to alleviate this is to adjust the dogbones of the holes in the plate and the dogbones around the tenons.
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/design.html#_inside_sharp_corners

[image]

karlrahn, 2022-11-21:
If I fix that, then I'm pretty sure the mounts won't move at all and the other question will be answered probably as well.

karlrahn, 2022-11-27:
Fixed

maidlam, 2022-11-27:
The tenons have more support now, but the dogbones for the mortises are unmillable - #45

closed 2022-11-27 by maidlam