ut-robotics/picr22-team-jackbot #34: [Mechanics] What's the process for attaching the components and opening up the robot?

maidlam, 2022-11-21:
The robot seems quite complex to assemble and open up. Getting inside the robot easily is a good functionality to have.
What are the steps for opening up the robot and which components will be attached to what after opening?

An issue I can see happening can be with the motor mounts. Because they are not attached with bolts to anything, they can come out of their place when removing one of the plates.

karlrahn, 2022-11-21:
The LeBot had similar connection for the motormounts and it seems that there is no problem. There are 2 holders, beacause the motors weight will balance on them 2 and will keep it static. There are several spacers between the lower and upper plate so there is no problem for the robot to fall apart. And if the upper plate is removed for some reason then I'm pretty sure that the motor with the mounts will stay in place.

karlrahn, 2022-11-27:
Can discuss it again tomorrow.