ut-robotics/picr22-team-jackbot #35: Don't place net labels upside down

ReikoR, 2022-11-27:
[image]
[image]

incrut, 2022-11-27:
[image]
Of course, I would not place the nets like that on purpose, and on my PC, it looks not turned upside down. However, for some reason, after I upload the file, it turns them this weird way.

ReikoR, 2022-11-27:
Are you using EasyEDA and converting to Altium Designer? If that's the case, then I would like to review using EasyEDA design files. There is no point of converting the files.

incrut, 2022-11-27:
> Are you using EasyEDA and converting to Altium Designer? If that's the case, then I would like to review using EasyEDA design files. There is no point in converting the files.

Oh yeah, sure, I just thought that you were using only Altium. Where should I send the EasyEDA file? To the electronics repository branch?

ReikoR, 2022-11-27:
Commit and push it to `electronics` branch and `electronics` folder.

closed 2022-11-29 by ReikoR