ut-robotics/picr22-team-jackbot #38: Add decoupling capacitors to microcontroller

ReikoR, 2022-11-27:
[image]

Each VDD/VDDA pin should have a decoupling capacitor (typically 100nF). One additional capacitor with higher capacitance (4.7uF is recommended in the datasheet) somewhere near MCU is also recommended.

ReikoR, 2022-11-29:
[image]

Decoupling capacitors should not be connected in series, but in parallel with VDD(A) and GND.

Also add one 4.7uF capacitor.

incrut, 2022-11-30:
> [image]
> 
> Decoupling capacitors should not be connected in series, but in parallel with VDD(A) and GND.
> 
> Also add one 4.7uF capacitor.

What should be the second leg connection for the capacitors, then? One leg is connected to the VDD(A) pin of the microcontroller, and the second leg is connected to the ground?

ReikoR, 2022-11-30:
Yes, capacitor should be between VDD(A) and GND.

> parallel with VDD(A) and GND

incrut, 2022-11-30:
[image]

> Yes, capacitor should be between VDD(A) and GND.
> 
> > parallel with VDD(A) and GND

Should it look like this in the end?

incrut, 2022-11-30:
> [image]
> 
> > Yes, capacitor should be between VDD(A) and GND.
> > > parallel with VDD(A) and GND
> 
> Should it look like this in the end?

 But with the 4.7uF capacitor

ReikoR, 2022-11-30:
Yes

closed 2022-11-30 by ReikoR