ut-robotics/picr22-team-jackbot #39: Missing battery input and motor driver output connectors

ReikoR, 2022-11-27:
[image]

Connectors or pads are needed for battery input and motor driver outputs. I recommend simple SMD pads where wires can be soldered.

ReikoR, 2022-11-29:
SMD pads should also be added for battery input.

Reverse voltage protection should also be added:
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/electronics/components.html#_reverse_voltage_protection

ReikoR, 2022-11-30:
1. BUK6U10-30P body diode has wrong direction.
[image]
\
From [datasheet](https://assets.nexperia.com/documents/data-sheet/BUK6Y10-30P.pdf):
[image]

2. I can't find BUK6U10-30P on the PCB. Battery input pad is connected straight to +16V.
[image]

incrut, 2022-11-30:
> 1. BUK6U10-30P body diode has wrong direction.
>    [image]
>    
>    From [datasheet](https://assets.nexperia.com/documents/data-sheet/BUK6Y10-30P.pdf):
>    [image]
> 2. I can't find BUK6U10-30P on the PCB. Battery input pad is connected straight to +16V.
>    [image]

It was the initial design where I have not added reverse voltage protection yet.

closed 2022-12-03 by ReikoR