ut-robotics/picr22-team-jackbot #46: No need for second support of motors

Akustav, 2022-11-27:
I suggest either removing the motor back supports or make the slots for them bigger so there is a loose fit. Otherwise it is a horror show to assemble since you need to align absolutely every slot with its tab. Use as few precise alignment faces as possible

[image]

karlrahn, 2022-11-27:
will do!

karlrahn, 2022-11-27:
Made the motor holder rear mount hole a bit wider

maidlam, 2022-11-29:
The motor support slots are currently 5 mm thick (vs old 4,8 mm), length is still the same. I think they should be bigger.
And I think the bottom plate's slots shouldn't be widened, only the slots in the top plate for the motor suports.

@Akustav How much increase would you suggest for the slots?

karlrahn, 2022-11-29:
I made both holes 0.5 mm wider now on the top plate, so that the bottom plate has 4.8mm

Akustav, 2022-11-30:
Around 0.5mm slop should be enough to make assembly simple and still locate them well enough

closed 2022-12-01 by Akustav